afpineda/OpenSourceSimWheelESP32 #4: HELP: Funky Problem - Push buttom

FRS78, 2023-12-15:
Hi all,

I have implemented Setup1 in my steering wheel project and everything works correctly except for the funky switch (RKJXT1F).

The problem is that when I connect the push button pin it generates a conflict and the funky stops working and only the push function works (and that function activates two things, the push and the crosshead up).
If I don't connect the push button pin, the funky works correctly (except the push button).

I have checked everything and it is correct (same way that the original project), but it does not work on my PCB or on my PROTOBOARD.

Please, has anyone had this problem and knows how to solve it?

[image]

FRS78, 2023-12-15:
Added full screenshoot
[image]

janc18, 2023-12-16:
Hi @FRS78, What is your test circuit?
I just see an issue with the S1-AZUL, at the datasheet say that is  the common push pin, but you are taking like another switch.

FRS78, 2023-12-16:
Hi @janc18 

I do not understand what you mean...

I have made a summary of how I have it connected. I think it is the same as the original project defines it.

If you see something strange, tell me because that may be the error and I won't see it.

Hope to solve this, because i expend so much time with this issue, jrjrjrjrjrjr

[image]
[image]

afpineda, 2023-12-19:
Hi, Fernando.

I will look into this issue as soon as I can, but it can take some time.
I have an ALPS enconder over here, so I can do some testing.

Don't discard a faulty hardware design, since *setup1* was not built for a full system test (just the firmware was tested).

afpineda, 2023-12-19:
Hello again, Fernando.

> I have been studying the topic and I already understand what you were referring to. It seems that the configuration in this funky switch the push button cannot be connected in any way that is an independent button... Can you confirm that this is the case???

I think there is a **big mistake** in the ALPS datasheet you posted before that is misleading you.

[image]

Try to swap `COM`  and `Push` terminals and tell us what happens.

janc18, 2023-12-19:
> I have been studying the topic and I already understand what you were referring to. It seems that the configuration in this funky switch the push button cannot be connected in any way that is an independent button... Can you confirm that this is the case???
> 

Hi @FRS78, yeah, that was exactly what I was trying to explain.
I will give the documentation a read to find an alternative.

FRS78, 2023-12-20:
Hi @janc18 and @afpineda

Confirmed, if you connect the push pin to gpio22 and the com pin to gpio4 everything works correctly. What's more, I have verified that the rest of the buttons work correctly, including the ALT buttons.

What remains pending is:
Option 1: Document this change in the original schematic
Option 2: edit the code

closed 2023-12-20 by FRS78

FRS78, 2023-12-20:
Post-data (spanish)

Muchas gracias por vuestra colaboración @janc18 / @afpineda 
Estaba atascado con esto y no veía la luz

Aprovecho para felicitaros por este proyecto porque es Super chulo, muy flexible y funcional
Una verdadera lástima no poder usar estas preciosas PCB´s que ya tengo hechas:

[image]

Lo que voy a hacer es editar mi PBC para tener la funcionalidad TOTAL del proyecto, por eso si se cambia el código por favor haced un post para reactualizar las PCB´s si hiciera falta (aunque lo más práctico es modificar el esquemático original del proyecto y fin del problema)

Saludos y muchas gracias por vuestra atención

janc18, 2023-12-20:
Hi @FRS78 and @afpineda.

If anyone has a problem in the future or needs help or feedback, don't hesitate to contact me. We have somewhat similar projects in the achievement of DIY equipment for simracing.

afpineda, 2023-12-21:
Hi.

> What remains pending is: Option 1: Document this change in the original schematic Option 2: edit the code

The mistake is at the ALPS datasheet. The code and the circuit design is not faulty, nor your printed PCB is. I feel sorry for you. 
This is the second time a misleading tag is causing an issue.